ergogen/ergogen #3: Example Config?

droppedoncaprica, 2020-10-18:
It would be really awesome if there could be an example configuration somewhere in the repository to view what the results would roughly look like.  You could then work from there and work on your own design and lean on the docs to add whatever you need from there.

Overall though, this is really cool!

mrzealot, 2020-10-18:
Hey, thanks for your interest! There's gonna be a small(er) yet representative default config in the repo (plus within the yaml box) to let people start playing around right away, but I think the Absolem is too complex for that. If it ever gets hard-copied into the ergogen repo, it will be accompanied with a verbose tutorial/annotation about how it came to be. But example curation is gonna be a larger task anyway.

For now, please revise your PR to just link to the original Absolem config from the ergogen docs if there's no such link yet, and we'll revisit the examplification after there's proper "on-boarding" tutorials and smaller examples already.

droppedoncaprica, 2020-10-18:
Yeah, that totally makes sense.  Updated the reference doc, let me know if you need any other changes.

mrzealot, 2020-10-18:
Cool stuff, I think it's mostly in the right place, and can be merged after a quick fix I suggested.

Also, since this is an example configs issue, let me hijack this thread to mention an idea from reddit: there should be a dropdown next to the input textarea where easy/easier pre-made configs for more well-known brands like Corne/Kyria/Atreus/Lily/Ergodox can be selected as starting points. They don't have to contain all bells and whistles, just enough to get people started more quickly. @droppedoncaprica I guess this would also serve as a solution to your original issue as well, wouldn't it?

droppedoncaprica, 2020-10-18:
Yep, that would fix the issue I think!  I like the idea of starting with an already established keyboard since really most people would probably like to just use something that's already there with a few tweaks.  Plus it would give them a good starting point for their own ideas.

As far as the PR, I'll make the change real quick and push it up.

mrzealot, 2020-11-02:
The dropdown is live, although with just two mini examples for now. But the infrastructure is in place, and the quantity can be increased as we go along. Guess this should close this, but feel free to reopen or post another issue if there's anything else.

closed 2020-11-02 by mrzealot