ergogen/ergogen #5: Error: Cannot find module 'fs-extra'

brow, 2020-10-31:
I'm trying to run the CLI by following these instructions:

> Command line usage requires `node v14.4.0+` with `npm v6.14.5+` to be installed, the repo to be checked out, and then simply calling the CLI interface through `node src/cli.js`.

Here is what happens:

``` bash
> git rev-parse HEAD
f4835e53151ce1a69a7b2c5d08bd6d7b3e60258c
> node --version
v14.5.0
> npm --version
6.14.5
> node src/cli.js
internal/modules/cjs/loader.js:1033
  throw err;
  ^

Error: Cannot find module 'fs-extra'
Require stack:
- /Users/tombrow/Documents/ergogen/src/cli.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:1030:15)
    at Function.Module._load (internal/modules/cjs/loader.js:899:27)
    at Module.require (internal/modules/cjs/loader.js:1090:19)
    at require (internal/modules/cjs/helpers.js:75:18)
    at Object.<anonymous> (/Users/tombrow/Documents/ergogen/src/cli.js:5:12)
    at Module._compile (internal/modules/cjs/loader.js:1201:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1221:10)
    at Module.load (internal/modules/cjs/loader.js:1050:32)
    at Function.Module._load (internal/modules/cjs/loader.js:938:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/Users/tombrow/Documents/ergogen/src/cli.js' ]
}
```

mrzealot, 2020-10-31:
I think you should issue `npm install` first. Honestly, it seems like a case of me just assuming that it's an "obvious" preparatory step, and forgetting to include it in the docs. Please retry that way, and if it's indeed that, I'll just add it to the docs.

brow, 2020-11-01:
Thanks, that did it! I'll leave this issue open in case you want to associate the doc change with it.

commit 9e2d30f0 (2020-11-01): Minor doc fix

Closes #5
closed 2020-11-01 by mrzealot