ergogen/ergogen #8: TypeError: Cannot read property '0' of null

brow, 2020-11-07:
I encounter this error when I [set my `glue`'s `top.right.rotate` equal to `points.rotate` and `top.left.rotate` equal to the negation of that](https://github.com/mrzealot/ergogen/commit/8f416d7e0d1e39a4b832cebc0ddcd024d460fb38).

``` bash
> git rev-parse HEAD
8f416d7e0d1e39a4b832cebc0ddcd024d460fb38
> node src/cli.js -c jklp.yaml -o jklp --debug
Parsing points...
Generating outlines...
/Users/tombrow/Documents/ergogen/src/utils.js:6
    return a[0] === b[0] && a[1] === b[1]
                     ^

TypeError: Cannot read property '0' of null
    at Object.exports.eq (/Users/tombrow/Documents/ergogen/src/utils.js:6:22)
    at /Users/tombrow/Documents/ergogen/src/outlines.js:155:27
    at Object.exports.parse (/Users/tombrow/Documents/ergogen/src/outlines.js:238:27)
    at Object.<anonymous> (/Users/tombrow/Documents/ergogen/src/cli.js:79:31)
    at Module._compile (internal/modules/cjs/loader.js:1201:30)
```

The error does not occur if `top.right.rotate` differs from `points.rotate` or from negative `top.left.rotate`.

mrzealot, 2020-11-08:
Got it, and it's sadly yet another corner case of [Maker.JS](https://github.com/microsoft/maker.js/blob/931ca948197d88362f6d404f24f5da141e694227/packages/maker.js/src/core/point.ts#L175). When the lines are parallel, it returns `null` instead of the halfway point. In your exact case, this can be circumvented if you take 3 large breaths, defeat your OCD, and change the anchor rotations to `24.99` instead of `25`. Otherwise, this would need yet another Maker.JS hotfix, which I would like to postpone if at all possible...

Anyhow, I committed a better error message for these cases. Feel free to reopen if this is insufficient!

closed 2020-11-08 by mrzealot

brow, 2020-11-08:
That sounds like a fine workaround. Thanks!