ergogen/ergogen #14: Error while parsing absolem.yaml using ergogen

warfareonly, 2021-02-18:
Hi, I was trying to generate the output for the default absolem board (with the cases & everything), so I tried to use the absolem config file from the absolem repo (from [here](https://github.com/mrzealot/absolem/blob/master/absolem.yaml)).

However, I get the following error:
```Javascript
Preparing input...
Parsing points...
/home/bharat/keyboards/ergogen/src/assert.js:12
        throw new Error(msg)
        ^

Error: Unexpected key "extends" within field "points.zones.s19"!
    at exports.assert (/home/bharat/keyboards/ergogen/src/assert.js:12:15)
    at Object.exports.detect_unexpected (/home/bharat/keyboards/ergogen/src/assert.js:35:9)
    at exports._render_zone (/home/bharat/keyboards/ergogen/src/points.js:22:7)
    at Object.exports.parse (/home/bharat/keyboards/ergogen/src/points.js:254:28)
    at Object.process (/home/bharat/keyboards/ergogen/src/ergogen.js:19:44)
    at Object.<anonymous> (/home/bharat/keyboards/ergogen/src/cli.js:66:25)
    at Module._compile (node:internal/modules/cjs/loader:1108:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1137:10)
    at Module.load (node:internal/modules/cjs/loader:973:32)
    at Function.Module._load (node:internal/modules/cjs/loader:813:14)
```

I'm running node v15.8.0 and npm v6.14.11 on Manjaro 5.10.15.

warfareonly, 2021-02-18:
Ah. also, if I use the same config file on the WebUI for the ergogen, I get a "SyntaxError: Unexpected token u in JSON at position 0" error. 
Unfortunately, my skill with JS is less than zero, otherwise I'd try to track down what is wrong myself :(

mrzealot, 2021-02-18:
Howdy! Thanks for catching this! Turns out my local Absolem config was out of sync with the repo... Pushed my updates, should compile now with the freshest ergogen version.

Also, I'm aware that the web UI is out of sync with the command line ergogen, but haven't had the time to update yet. Please use the command line for now as there have been breaking changes (which are still pretty much un-documented). I hope I can get around to a little organization soon...

closed 2021-02-18 by mrzealot