ergogen/ergogen #33: `affect: xyr` on the mirrored side doesn’

torik42, 2021-09-10:


MvEerd, 2021-09-10:
I believe part of the issue here is that you're defining two anchors here, one of them being
```  
- ref: mirror_matrix_only_top
   affect: xyr 
``` 
the other 
```
- shift: [-3,-3]
  orient: 30
 ``` 
 
 Did you intend for the `shift` to be relative to your mirror, and be affected by your `affect` param? Then these should be under the same anchor like so
  ``` 
         anchor:
          - ref: mirror_matrix_only_top
            affect: xyr
          - shift: [-3,-3]
            orient: 30
  ``` 
  
 When `affect: xyr` is used without using 'shift' in the same anchor it will not mirror the xyval first, resulting in different behaviour.
 Perhaps `affect` should not be allowed without either `shift` or `rotate`?

torik42, 2021-09-10:
My intention was to place a rotated rectangle in the middle of the key. Honestly, I don’t understand the relative-to-the-mirror-magic properly.
Since rectangles are placed according to their lower left corner, I needed to use orient and shift. My first guess was
```
      - type: rectangle
        anchor:
          ref: mirror_matrix_only_top
          shift: [-3,-3]
          orient: 20
        size: [6,6]
```
which works on the non-mirrored side, but not on the mirrored one:
[image]

For non-rotated keys, one can use
```
      - type: rectangle
        anchor:
          - ref: mirror_matrix_only_toptop
            affect: xy
          - shift: [-3,-3]
            orient: 30
        size: [6,6]
```
but I just thought, what happens with rotated keys:
[image]
Added is the first row with `affect: xy` (second is `affect: xyz`, third is without) and of course the rotated keys. I don’t know how useful this all is.

As far as I now understand, if I have a ref from the mirror, then also shifts (but not rotations, nor orients?) are mirrored, such that the result looks like a mirror image. But since the placement of rectangles is also not mirrored, I need to break this mirroring. Maybe one should also mirror orients, rotations and the placement of rectangles?

MvEerd, 2021-09-10:
How are you defining the anchors for all these?
What do you expect your example to do here?
```
      - type: rectangle
        anchor:
          - ref: mirror_matrix_only_toptop
            affect: xy
          - shift: [-3,-3]
            orient: 30
        size: [6,6]
```

Here you say, anchor this rectangle to `mirror_matrix_only_toptop`, only for x and y
and also anchor this to no ref and shift and orient it based on the default_point

Are you trying to shift and orient it based on the ref? then the `-` shouldn't be there in front of `shift`

Does this give the result you expect? 
If so I'm not sure this is a bug, but that the `x` in shift should be negated if you're anchoring it to a mirrored point?
```
      - type: rectangle
        anchor:
          - ref: mirror_matrix_only_bottom
            affect: xyr
            shift: [3,-3]
            orient: 30
        size: [6,6]
        operation: stack
      - type: rectangle
        anchor:
          - ref: mirror_matrix_only_top
            affect: xyr
            shift: [3,-3]
            orient: 30
        size: [6,6]
      - type: rectangle
        anchor:
          - ref: matrix_only_bottom
            affect: xyr
            shift: [-3,-3]
            orient: 20
        size: [6,6]
        operation: stack
      - type: rectangle
        anchor:
          - ref: matrix_only_top
            affect: xyr
            shift: [-3,-3]
            orient: 20
        size: [6,6]
        operation: stack
 ```

torik42, 2021-09-10:
As far as I understood, one can chain these things. And with no given ref, the last point is used as a starting point.

MvEerd, 2021-09-10:
This is not how I'm interpreting the following lines in the doc/reference
> an omitted ref in the anchor means the current key for key-level declarations, while here it defaults to [0, 0]

> Also note that anywhere an anchor can be specified, a list of anchors is also valid. It would be meaningless, though, if each subsequent anchor would override the previous one, so the affects clause helps to affect only certain dimensions of the anchor. It can be declared using a string containing any of x, y, or r, which stand for the x and y coordinates and the rotation of the anchor, respectively.

From the above I expect your second anchor without ref to overwrite the first one, unless you omit x y or r from either of them

torik42, 2021-09-11:
I think, I now understood the problem and have a few ideas what to do. Will create a Pull request soon.

mrzealot, 2021-12-12:
Partially fixed the `affect` inconsistency, but full visual symmetry will have to wait until we lay out rectangles by their centers instead of their bottom left corners, which (as a breaking change) will be a separate issue, and is now already on the TODO. Anyhow, thanks for the investigation! :heart:

commit 38a8fc5b (2021-12-12): Anchor affect bugfix, partially fixes #33

The mirroring part still has design inconsistencies because of rectangles being laid out by their bottom left corner, but as a breaking change, this will be addressed separately, later...
closed 2021-12-12 by mrzealot