ergogen/ergogen #42: Use a 2u key

toniz4, 2021-10-19:
It's possible to set a single key to 2u? I want to make something like the corne, that has a 2u key as the outer thumb key, but I didn't find how to do that. 

Thanks for developing this awesome program!

Albert-IV, 2021-11-07:
I believe all you would have to do to support that would be rotate the key by 90°.

```
              choc:
                type: choc
                anchor:
                  rotate: 90
                nets:
                  from: =colrow
                  to: =column_net
                params:
                    keycaps: true
                    reverse: true```

mrzealot, 2021-12-12:
@Albert-IV is right! There's not much support for keycaps, so if you need to leave exact space for the 2u keycap then you need to operate a bit with rectangles, but as far as the switch itself goes, you can just place it in the middle by some `shift`ing, and probably rotate by 90° if you want it to be "tall" instead of "wide".

closed 2021-12-12 by mrzealot