ergogen/ergogen #53: Best way to implement per-key LEDs

rickyelopez, 2022-03-16:
I'm using this tool for the first time and loving it! So far, I've been able to implement the few things that the tool couldn't do on my own, but I'm struggling with this. I've been looking through the source code and I don't see an immediately obvious way of implementing this nicely, was hoping someone here might have a suggestion.

I'd like to implement per-key LEDs for a keyboard I'm working on. I already have the LEDs getting placed, but I'm struggling to come up with a way to create and link the nets. The LEDs I'm using are `sk6812`s, which have VCC and GND (easy) as well as DIN and DOUT, which need to be linked to the previous and next LED in the chain, respectively. The first LED needs to have DIN linked to a pin on the micro, and the last LED leaves DOUT unconnected.

There are two problems here:
1. need to be able to traverse the list of keys in order to link each LED to either the previous or next LED (probably next)
2. there needs to be some way to define the chain in the yaml.

The first problem is mostly just due to unfamiliarity with the source code. There's probably an easy way to do this.

As for the second one, presumably every layout will have a different ideal chain pattern. For example, you might want to start at the top left and zig-zag across the rows, or you might want to go column-by-column, etc. I can't think of a good way to define this chain in the yaml file.

Anybody have any suggestions or opinions on what a good way to implement this might be? Thanks!

mrzealot, 2022-03-16:
Hmm, there's no "easy" or "officially supported" way to do this - tbh, this hasn't come up yet. Also, this is a declarative config language, so no list support, no traversal. But if I wanted to do this, I'd probably do it via key-level attributes. Not as convenient, but you could add a `led_next` and `led_prev` attribute to every key in the matrix with the appropriate net names, which for the first would be the controller pin, last would be empty, and the middle ones could be `led_1` through `led_n`. If this is set, you can then use `=led_prev` and `=led_next` within the footprint to reference the key-level attributes and everything gets properly connected, theoretically.

Also note that there's nothing stopping you from programmatically generating your yaml config if you need procedural help. Ergogen can just be a middle layer if that's more convenient. You write code to generate your keyboard, only you don't have to mess with the nuts-and-bolts, just output an ergogen config and let it do the rest.

Also also, I'd advise you to come discuss these kinds of things on [Discord](https://discord.gg/nbKcAZB) where others often have better ideas than I do :joy:

rickyelopez, 2022-03-16:
> do it via key-level attributes. Not as convenient, but you could add a led_next and led_prev attribute to every key in the matrix

haha yeah this is exactly what I was trying to avoid! Good point about programmatically generating the yaml, but honestly this seems to me like something that other people might want to do. I'm surprised this hasn't already come up tbh.

Good idea re: joining the discord, will do tonight! Thanks for the feedback

ceoloide, 2023-06-06:
For people stumbling on this post v4, here's how you can add key level attributes like this:

```
zones:
  matrix:
    columns:
      inner:
        rows.top.key.led_prev: LED
```

and then you can refer to the additinal attribute in the footprint as `{{key.led_prev}}`

drelephant, 2024-02-21:
@ceoloide Any chance you could post a minimal board `yaml` example using this?

For the people like me who aren't experienced enough to work out how to use that?

ceoloide, 2024-02-21:
> @ceoloide Any chance you could post a minimal board `yaml` example using this?
> 
> For the people like me who aren't experienced enough to work out how to use that?

Sure, check out my [Mr. Useful "macropad"](https://github.com/ceoloide/mr_useful): it has both underglow and per-key LEDs, and it's a simple one-sided board. If you want something more complicated, or a reversible example, you can check my Corne clone, the [Corney Island](https://github.com/ceoloide/corney-island).

Hope these are useful to you!