ergogen/ergogen #74: Source code for the 'unofficial' (ergogen.cache.works) webUI

mhk, 2023-01-10:
Since 4.0.0 introduced a lot of breaking changes a lot of older configurations are not working anymore. For example the Atreus (simplified) on https://ergogen.cache.works/.
Since the 'unofficial' webUI, aka the one used on https://ergogen.cache.works/ is very useful for development, it would be helpful to still use the old configs.
There are two solutions that come to mind:
1.  publicly host a version that accepts the old configs
2. get access to the source code (https://ergogen.cache.works) to host it locally

l-kershaw, 2023-01-11:
The unofficial webUI is in a repo [here](https://github.com/MvEerd/ergogen-gui)

mhk, 2023-01-12:
Thanks a lot!

closed 2023-01-12 by mhk