ergogen/ergogen #95: Just a few thoughts

nine-fox, 2023-05-19:
Hi there,

This is really a cool project! Lots of features are fascinating!

Coincidentally, I was doing the same thing in the last one year. A different route to help DIYers to create keyboard automatically.

[image]

You may take a look at:
https://nine-fox.com/diy

While a simple guide locates:
https://nine-fox.com/docs/diy/#/quickstart

screenshots (Chinese page although):
https://www.zfrontier.com/app/flow/9zQGMWLLkVg8

So this is my different approach and my thoughts are as follows you may take consideration if you like.

1) KiCAD vs JLCPCB

I was thinking to generate the whole sch and pcb using KICAD, however, the main drawback is, it needs lots professional skills and also needs to download and install it from website(what a big package!). And, there is no great support for SMT directly even with skills.

To me, I didn't like handsoldering, and searching every pieces of components from taobao or market. That's really annoying and cost lots time. So I moved to gen PCB under JLCPCB(https://lceda.cn, https://jlcpcb.com/easyeda). Currently, I can gen rectangle keys(probably support custom rotation in the future).

2) 2D vs 3D 

With the help of https://github.com/zalo/CascadeStudio, opencascade wasm can generate the everthing in 3d format(glb and STEP). While Glb can be used for 3d printing, however, to me, I think the most importance is to generate STEP while CNC is supported. That would be a great help for opening group buy. I would suggest to support it. There.js's open source editor would help.

3) Realistic rendering

I would strongly suggest to support 3d realistic rendering. That's a quite great feature to view the best result.
[image]

While keycap can be generated, its not difficult to render online with webgl. There are lots of open sourced projects(https://github.com/topics/path-tracing). You may take a look at the rendering screenshot my website at:
https://www.zfrontier.com/app/flow/9zQGMWLLkVg8

By the way, if you want to download the whole zip, you may reach me to get the free code. Any thoughts are welcomed.

98a7875a21dc

That's download code today.

mrzealot, 2023-05-20:
Hey @nine-fox ,

Well shit, did I get upstaged or what! :eyes: This tool looks fantastic! Not really a *direct* competitor on a "design philosophy" level, as far as I can judge by a quick look, but Jesus, there's a lot Ergogen could learn from you.

Re: 1, I've been very set on kicad, and plan to continue supporting it as the main PCB option, but Ergogen could probably benefit from looking into EasyEDA as an alternative.

Re: 2, There were plans already to significantly enhance the 3D aspects of the tool once the documentation is finally complete and up-to-date, but those plans will absolutely expand thanks to this round of inspiration. Originally, I was going to go with CadQuery, but I'll take opencascade (and the studio it comes with! :eyes: ) into consideration.

Re: 3, now this I haven't even thought about yet. But not gonna lie, it looks great, adds a lot to the experience, and now I'm definitely going to plan for adding it eventually.

My biggest problems with all of the above is the VERY limited time I can afford to contribute to active development - and even that much has been going towards proper testing and documentation for a while now :( But this is a very eye-opening kick in the ass, thank you! :joy: I'll do my best to steer development to match this level of usability while retaining the exact, declarative nature of the input configs.

Definitely interested in the source code, if you don't mind. (Probably late for the above download code, tho... And would also appreciate a bit more direction in how to obtain the codebase)

Oh, and also: do join us at discord.ergogen.xyz if you feel like discussing this further :+1:

nine-fox, 2023-05-31:
> Hey @nine-fox ,
> 
> Well shit, did I get upstaged or what! 👀 This tool looks fantastic! Not really a _direct_ competitor on a "design philosophy" level, as far as I can judge by a quick look, but Jesus, there's a lot Ergogen could learn from you.
> 
> Re: 1, I've been very set on kicad, and plan to continue supporting it as the main PCB option, but Ergogen could probably benefit from looking into EasyEDA as an alternative.
> 
> Re: 2, There were plans already to significantly enhance the 3D aspects of the tool once the documentation is finally complete and up-to-date, but those plans will absolutely expand thanks to this round of inspiration. Originally, I was going to go with CadQuery, but I'll take opencascade (and the studio it comes with! 👀 ) into consideration.
> 
> Re: 3, now this I haven't even thought about yet. But not gonna lie, it looks great, adds a lot to the experience, and now I'm definitely going to plan for adding it eventually.
> 
> My biggest problems with all of the above is the VERY limited time I can afford to contribute to active development - and even that much has been going towards proper testing and documentation for a while now :( But this is a very eye-opening kick in the ass, thank you! 😂 I'll do my best to steer development to match this level of usability while retaining the exact, declarative nature of the input configs.
> 
> Definitely interested in the source code, if you don't mind. (Probably late for the above download code, tho... And would also appreciate a bit more direction in how to obtain the codebase)
> 
> Oh, and also: do join us at discord.ergogen.xyz if you feel like discussing this further 👍

Hi @mrzealot , sorry for the late reply. Been busy these days and China GFW does not allow me to access github occasionally.

You are right, its a "design philosophy". In short, I am WEB Usability oriented and you are PROS & SKILLS oriented :-)

You mentioned the time issue as your biggest problem, hmmm, right, mine is the same. I have job too, and time is so limited, and I can only develop it in spare time. But passion is more important, we are doing the thing that we loved, which means a lot.

I did a lot investigation on CadQuery, which is really a great tool and quite good documentation. It just stopped me since I am not so familiar with python, and also, it cannot be used on WEB.

For the code base, there are lots of path tracing projects already. I would recommend https://github.com/gkjohnson/three-gpu-pathtracer as its well documented. And there's another project is great too: https://erichlof.github.io/THREE.js-PathTracing-Renderer/Geometry_Showcase.html

By the way, I am afraid I can not access discord due to China GFW.  Opps, wish we can discuss further using another channel.

You're doing great! I will keep an eye on this project~

nine-fox

closed 2023-05-31 by nine-fox