ergogen/ergogen #98: STL export not working

Nisker, 2023-06-16:
Exporting STL from both the official and unofficial site gives me a 9 bytes file.

It can be replicated by generating an STL of CasePcbTest on https://ergogen.xyz/

kitnerboy, 2024-02-05:
Can confirm. I see the same behaviour: 9 byte stl with single string in it "undefined".

TildeWill, 2024-02-10:
A workaround for now is to pipe the *.jscad file through a node package to get an *.stl. FlatFootFox outlines more in their tutorial in the section labeled "Making A Case" https://flatfootfox.com/ergogen-part4-footprints-cases/

Ermakov-code, 2025-09-02:
Has anyone encountered a situation where an STL file looks strange in the preview and cannot be loaded into any STL online viewer because it is supposedly corrupted?

closed 2026-06-25 by ceoloide