ergogen/ergogen #101: Error: Field "outlines.exports.raw" should be of type object!

caldayham, 2023-07-15:
I am trying to gen my new keyboard but I keep getting the error that 'raw' should be an object, it looks like a list, but when I remove the dashes the error changes to be 'unknown attribute "type"'. I am using basically the same outlines as benvallack who does not seem to be having any issues. I am very new to ergogen.

here is my current file:
`points:
  key:
    padding: cy
    tags: 
      1u: true
    footprints:
      choc_hotswap:
        type: choc
        nets:
          from: =column_net
          to: GND
        params:
          keycaps: true
          reverse: true
          hotswap: true
      choc:
        type: choc
        anchor:
          rotate: 180
        nets:
          from: =column_net
          to: GND
        params:
          keycaps: true
          reverse: true
  zones:
    mousemode:
      anchor:
        rotate: 0
        shift: [94, -5]
    pinkygroup:
      anchor:
        rotate: 10
        shift: [-22, 16]
      columns:
        pinky:
      rows:
        top:
        bottom:
    matrix:
      anchor:
        rotate: 10
      columns:
        pinky:
        ring:
          key.origin: [-12, -19]
          key.stagger: 8
          key.splay: -4
        middle:
          key.stagger: 2
          key.splay: -4
        index:
          key.stagger: -2
          key.splay: -4
        inner:
          key.stagger: -8                     
      rows:
        bottom:
        home:
        top:
    thumbgroup:
      anchor:
        ref: matrix_inner_bottom
        shift: [-32, -32]
      columns:
        inner:
          key.splay: -22
        home:
          key.splay: -4
        outer:
          key.splay: -4
        far:
          key.splay: -4
      rows:
        upper:
        lower:
  rotate: -20
outlines:
  exports:
    raw: 
      - type: keys
        side: left
        size: [1cx , 1cx]
      - type: rectangle
        size: [7, 5]
        anchor:
          ref: board_index_top
          shift: [21, 6]
    cutouta:
      - type: outline
        name: raw
        fillet: 3
    cutoutb:
      - type: outline
        name: cutouta
        fillet: 1
    cutout:
      - type: outline
        name: cutoutb
        fillet: 0.5
    keycap_outlines:
      - type: keys
        side: left
        size: [1cx - 0.5, 1cy - 0.5] # Choc keycaps are 17.5 x 16.5
        bound: false               
pcbs:
  card:
    outlines:
      main:
        outline: cutout
    footprints:
      promicro:
        type: promicro
        anchor:
          ref: board_far_bottom
          shift: [0.6, cx+6.3]
          rotate: -90
        params:
          orientation: down
      pcm12:
        type: pcm12
        anchor:
          ref: board_index_top
          shift: [16.5, 6.5]
          rotate: 180
        nets:
          from: pos  
          to: RAW
        params:
          reverse: true 
      via3:
        type: via
        anchor: 
          ref: board_index_top
          shift: [16,3]
        nets: 
          net: RAW 
      via4:
        type: via
        anchor: 
          ref: board_index_top
          shift: [18,3]
        nets: 
          net: pos 
      via5:
        type: via
        anchor: 
          ref: board_index_top
          shift: [23,6.5]
        nets: 
          net: GND 
      via6:
        type: via
        anchor: 
          ref: board_index_top
          shift: [25,6.5]
        nets: 
          net: RST 
      b3u1000p:
        type: b3u1000p
        nets:
          r1: RST  
          r2: GND
        anchor:
          ref: board_index_top
          shift: [24.5, 9]
          rotate: 0
        params:
          reverse: true 
      bat:
        type: bat
        nets: 
          neg: GND
        anchor:
          ref: board_far_bottom
          shift: [0.2 , 9]
          rotate: 90`

caldayham, 2023-07-15:
sorry, code got scrunched

points:
  key:
    padding: cy
    tags: 
      1u: true
    footprints:
      choc_hotswap:
        type: choc
        nets:
          from: =column_net
          to: GND
        params:
          keycaps: true
          reverse: true
          hotswap: true
      choc:
        type: choc
        anchor:
          rotate: 180
        nets:
          from: =column_net
          to: GND
        params:
          keycaps: true
          reverse: true
  zones:
    mousemode:
      anchor:
        rotate: 0
        shift: [94, -5]
    pinkygroup:
      anchor:
        rotate: 10
        shift: [-22, 16]
      columns:
        pinky:
      rows:
        top:
        bottom:
    matrix:
      anchor:
        rotate: 10
      columns:
        pinky:
        ring:
          key.origin: [-12, -19]
          key.stagger: 8
          key.splay: -4
        middle:
          key.stagger: 2
          key.splay: -4
        index:
          key.stagger: -2
          key.splay: -4
        inner:
          key.stagger: -8                     
      rows:
        bottom:
        home:
        top:
    thumbgroup:
      anchor:
        ref: matrix_inner_bottom
        shift: [-32, -32]
      columns:
        inner:
          key.splay: -22
        home:
          key.splay: -4
        outer:
          key.splay: -4
        far:
          key.splay: -4
      rows:
        upper:
        lower:
  rotate: -20
outlines:
  exports:
    raw: 
      - type: keys
        side: left
        size: [1cx , 1cx]
      - type: rectangle
        size: [7, 5]
        anchor:
          ref: board_index_top
          shift: [21, 6]
    cutouta:
      - type: outline
        name: raw
        fillet: 3
    cutoutb:
      - type: outline
        name: cutouta
        fillet: 1
    cutout:
      - type: outline
        name: cutoutb
        fillet: 0.5
    keycap_outlines:
      - type: keys
        side: left
        size: [1cx - 0.5, 1cy - 0.5] # Choc keycaps are 17.5 x 16.5
        bound: false               
pcbs:
  card:
    outlines:
      main:
        outline: cutout
    footprints:
      promicro:
        type: promicro
        anchor:
          ref: board_far_bottom
          shift: [0.6, cx+6.3]
          rotate: -90
        params:
          orientation: down
      pcm12:
        type: pcm12
        anchor:
          ref: board_index_top
          shift: [16.5, 6.5]
          rotate: 180
        nets:
          from: pos  
          to: RAW
        params:
          reverse: true 
      via3:
        type: via
        anchor: 
          ref: board_index_top
          shift: [16,3]
        nets: 
          net: RAW 
      via4:
        type: via
        anchor: 
          ref: board_index_top
          shift: [18,3]
        nets: 
          net: pos 
      via5:
        type: via
        anchor: 
          ref: board_index_top
          shift: [23,6.5]
        nets: 
          net: GND 
      via6:
        type: via
        anchor: 
          ref: board_index_top
          shift: [25,6.5]
        nets: 
          net: RST 
      b3u1000p:
        type: b3u1000p
        nets:
          r1: RST  
          r2: GND
        anchor:
          ref: board_index_top
          shift: [24.5, 9]
          rotate: 0
        params:
          reverse: true 
      bat:
        type: bat
        nets: 
          neg: GND
        anchor:
          ref: board_far_bottom
          shift: [0.2 , 9]
          rotate: 90

caldayham, 2023-07-15:
the indents are fine, i guess github removed them on paste

mrzealot, 2023-07-15:
Hiya! Haven't checked your config, but you mentioning Ben Vallack makes me strongly suspect that you're using v3 syntax (which was current when his vid came out), but we've released v4 since then with a bunch of breaking changes. The in-progress docs are getting better by the day, which you can check [here](https://deploy-preview-12--ergogen.netlify.app/), and there's also a dedicated community [migration guide](https://github.com/tsteffek/Ergogen-V4-Migration-Guide). Please try these, and if your question/confusion persists, feel free to ask in our [Discord](http://discord.ergogen.xyz)! :+1:

closed 2023-07-15 by mrzealot