ergogen/ergogen #140: Outlines and splay give unexpected result

doenietzomoeilijk, 2024-09-08:
I'm generating a layoout with splay on the ring and pinky column, together with an outline for it. This works, mostly, but there's a weird bump in the outline on the splayed columns. 

I think I somewhat understand *why* they happen; the bump is right above the topmost point and "jumps" the angle and distance between the pre and post splay column. If I add a fillet and round, it'll get filleted and rounded. However, I would like to somehow work around this behaviour and not "jump" the outline splay above the point, but *between* the points; this is where I'd also draw it if I were to do this manually.

When testing this, things worked as expected with one row of keys, the behaviour changed with a second row of keys.

I hope there's a way of working around this, otherwise maybe the behaviour could be mentioned in the documentation or possibly even modified. Again: I think I know what's happening, and it makes sense, in a way, it's just not what I'd expect from a plate-making user perspective.

---

Example YAML:

```yaml
---
points:
  zones:
    matrix:
      columns:
        pinky.key:
          splay: 4
          stagger: 0
        ring.key:
          splay: -2
          stagger: 0.5 * u
        middle.key:
          splay: -2
          stagger: 0.5 * u
      rows:
        bottom:
        middle:

outlines:
  holes:
    - where: true
      what: rectangle
      size: 14
  edge:
    - where: true
      what: rectangle
      bound: true
      size: u + 5
  plate:
    - edge
    - -holes
```

Resulting plate:

[image]

doenietzomoeilijk, 2024-09-08:
Huh, interesting, I've just changed the stagger on the two columns from 0.5 for both to 0.4 and 0.3, and poof, gone are the silly jumps. So at least that's one workaround, don't have extreme stagger. ;) Might also explain why this doesn't crop up for a lot of folks.

jrussellsmyth, 2024-12-22:
Actually, its doing exactly what you ask it to - just that binding is pretty complicated (I wont claim to fully understand it yet but..)

What is happening, because you only have two rows, the default autobind of "10" is making pinky_top and ring_top bind to the ring and middle bottom borders, and it looks like the algorithm makes that attempt to align like corners - note the horizontal on the  "bump" in the bottom middle points to the next corner.. same is happening on the top the other way around.

you can fix this by setting a smaller autobind, or by setting individual binds on the keys. try

```yaml
  points:
    key.autobind: 1
````