ergogen/ergogen #142: Outline disappears for one key

choyuansu, 2024-10-19:
As shown in screenshot below, one key is missing part of the outline
[image]

To reproduce:
<details>
<summary>YAML</summary>

```
units:
  # base units
  # MX spacing: U (19.05mm), u (19mm)
  # Choc spacing: cx (18mm), cy (17mm)
  kx: u
  ky: u

  # padding
  px: kx + 4
  py: ky + 4

  # double padding
  dpx: px + 4
  dpy: py + 4

points:
  zones:
    matrix:
      columns:
        pinky:
        ring.key.stagger: 0.66ky
        middle.key.stagger: 0.25ky
        index.key.stagger: -0.25ky
        inner.key.stagger: -0.15ky
      rows:
        bottom.padding: kx
        home.padding: kx
        top.padding: kx
        num.padding: kx
    thumb:
      anchor:
        ref: matrix_index_bottom
        shift: [0.66kx, -1.25ky]
        rotate: -10
      columns:
        tucky:
          key.name: thumb_tucky
        reachy:
          key.spread: kx
          key.splay: -15
          key.origin: [-0.5kx, -0.5ky]
          key.name: thumb_reachy

pcbs:
  simple_split:
    footprints:
      keys:
        what: mx
        where: true
        params:
          from: GND
          to: "{{name}}"
          reverse: true
          keycaps: true
      mcu:
        what: promicro
        where:
          - ref: matrix_inner_home
            shift: [1kx, 0.5ky]
            rotate: -90
        params:
          P7: matrix_pinky_top
          P18: matrix_ring_top
          P19: matrix_middle_top
          P20: matrix_index_top
          P21: matrix_inner_top
          P15: matrix_pinky_home
          P14: matrix_ring_home
          P16: matrix_middle_home
          P10: matrix_index_home
          P1: matrix_inner_home
          P2: matrix_pinky_bottom
          P3: matrix_ring_bottom
          P4: matrix_middle_bottom
          P5: matrix_index_bottom
          P6: matrix_inner_bottom
          P8: thumb_tucky
          P9: thumb_reachy

outlines:
  raw:
    - what: rectangle
      where: true
      size: [kx, ky]
      bound: true
  keys:
    - what: rectangle
      where: true
      size: [kx - 0.5, ky - 0.5]
  holes:
    - what: circle
      where: true
      radius: 1.5
      adjust:
        - shift: [-3.81, 2.54]
    - what: circle
      where: true
      radius: 1.5
      adjust:
        - shift: [2.54, 5.08]
    - what: circle
      where: true
      radius: 1.99
    - what: circle
      where: true
      radius: 0.85
      adjust:
        - shift: [5.08, 0]
    - what: circle
      where: true
      radius: 0.85
      adjust:
        - shift: [-5.08, 0]
  socket-holder-top:
    - what: rectangle
      where: true
      size: [12.9, 4]
      adjust:
        - shift: [-1, 6.755]
    - what: rectangle
      operation: subtract
      where: true
      size: [12.9, 4]
      corner: 2
      adjust:
        - shift: [-1+2, 6.755-2]
  socket-holder-bottom:
    - what: rectangle
      where: true
      size: [12.9, 4]
      corner: 2
      adjust:
        - shift: [-1+2+3.5, 6.755-2-4]
    - what: circle
      operation: subtract
      where: true
      radius: 1.99
    # issue here
    - what: rectangle
      operation: subtract
      where: true
      size: [5.5, 4]
      adjust:
        - shift: [-1+2+3.5+3.7, 6.755-2-4]
    - what: rectangle
      operation: subtract
      where: true
      size: [10.9, 0.865 * 2 + 2]
      adjust:
        - shift: [0, -1]

  board:
    - what: polygon
      operation: stack
      points:
        - ref: matrix_pinky_num
          shift: [-0.5px, 0.5py]
        - ref: matrix_ring_num
          shift: [-0.5px, 0.5py]
        - ref: matrix_middle_num
          shift: [-0.5px, 0.5py]
        - ref: matrix_middle_num
          shift: [0.5px, 0.5py]
        - ref: matrix_inner_num
          shift: [0.5px, 0.5py]
        - ref: matrix_inner_home
          shift: [0.5px, 0.5py]
        - ref: matrix_inner_home
          shift: [1.5px, 0.5py]
        - ref: matrix_inner_bottom
          shift: [1.5px, -1py]
        - ref: thumb_reachy
          shift: [0.5px, -0.5py]
        - ref: matrix_pinky_bottom
          shift: [0.5px, -0.5py]
        - ref: matrix_pinky_bottom
          shift: [-0.5px, -0.5py]
  xlBoard:
    - what: polygon
      operation: stack
      points:
        - ref: matrix_pinky_num
          shift: [-0.5dpx, 0.5dpy]
        - ref: matrix_ring_num
          shift: [-0.5dpx, 0.5dpy]
        - ref: matrix_middle_num
          shift: [-0.5dpx, 0.5dpy]
        - ref: matrix_middle_num
          shift: [0.5dpx, 0.5dpy]
        - ref: matrix_inner_num
          shift: [0.5dpx, 0.5dpy]
        - ref: matrix_inner_home
          shift: [0.5dpx, 0.5dpy]
        - ref: matrix_inner_home
          shift: [1.5dpx, 0.5dpy]
        - ref: matrix_inner_bottom
          shift: [1.5dpx, -1dpy]
        - ref: thumb_reachy
          shift: [0.5dpx, -0.5dpy]
        - ref: matrix_pinky_bottom
          shift: [0.5dpx, -0.5dpy]
        - ref: matrix_pinky_bottom
          shift: [-0.5dpx, -0.5dpy]
  combo:
    - name: board
    - operation: subtract
      name: keys
    - operation: add
      name: holes
  socket-holder: 
    - operation: add
      name: socket-holder-top
    - operation: add
      name: socket-holder-bottom

cases:
  bottom:
    - what: outline
      name: board
      extrude: 1
  xlBottom:
    - what: outline
      name: xlBoard
      extrude: 1
  _outerWall:
    - name: xlBoard
      extrude: 4
  _innerWall:
    - name: board
      extrude: 4
  wall:
    - what: case
      name: _outerWall
      operation: add
    - what: case
      name: _innerWall
      operation: subtract
  case:
    - what: case
      name: xlBottom
      operation: add
    - what: case
      name: wall
      operation: add
  socket-holder:
    - what: outline
      name: socket-holder
      extrude: 2
```

</details>

Preview "socket-holder.dxf."

Seems to only happen when I add the rectangle to subtract on lines 139-144

I expect the outline to show up normally like the other keys.

Any help is appreciated!

photomultiplier, 2024-10-20:
This can be fixed by increasing the size of the cutting rectangle (the one with the "issue here" comment) from `[5.5,4]` to `[5.5,5]`. Otherwise the top side of the cutting rectangle is aligned with the other rectangle... this causes numerical imprecision leading to buggy outlines.

choyuansu, 2024-10-20:
That works! Thank you so much!

closed 2024-10-20 by choyuansu