ergogen/ergogen #145: Using variable in points.zones.<zone_name>.columns.<col_name>.rows.mirror.width results in no image

jaybuilds222, 2025-01-01:
This works ok when using a number in this field:
[image]

Using a variable here results in no output:
[image]

jaybuilds222, 2025-01-01:
`
points:
  zones:
    matrix:
      columns:
        outer:
          rows:
            bottom:
              width: 2U - 1.05
              mirror:
                width: 18
            home:
        pinky:
          key:
            spread: 1.5U
        ring:
        middle:
      rows:
        bottom:
        home:
  mirror:
    ref: matrix_middle_home
    distance: 50
`

jaybuilds222, 2025-01-01:
```
points:
  zones:
    matrix:
      columns:
        outer:
          rows:
            bottom:
              width: 2U - 1.05
              mirror:
                width: 18
            home:
        pinky:
          key:
            spread: 1.5U
        ring:
        middle:
      rows:
        bottom:
        home:
  mirror:
    ref: matrix_middle_home
    distance: 50

```

ceoloide, 2025-01-03:
Here's another simplified example:

```yaml
points:
  zones:
    matrix:
      columns:
        col:
          rows:
            row:
              width: u
              mirror.width: 2u
  mirror:
    ref: matrix_col_row
    distance: 40
```

As far as I can tell, the mirrored key doesn't have a computed width, instead it maintains it as a string. Here is a condensed `points.yaml` output:

```yaml
matrix_col_row:
  x: 0
  'y': 0
  r: 0
  meta:
    width: 19
    name: matrix_col_row
mirror_matrix_col_row:
  x: 40
  'y': 0
  r: -0.0
  meta:
    width: 2u
    name: mirror_matrix_col_row
```

My bet is that the DXF / SVG conversion fails to display the result when it encounters `width: 2u` in the mirrored key.

ceoloide, 2025-01-03:
After running locally I can confirm that the SVG ends up with `NaN` where `2u` would be expected to be converted to `38`.

ceoloide, 2025-01-03:
In `points.js` the `visualize` function just reads the `width` and `height` off of the key `meta`, which for mirrored points will cause the issue.

```js
exports.visualize = (points, units) => {
    const models = {}
    for (const [pname, p] of Object.entries(points)) {
        const w = p.meta.width
        const h = p.meta.height
        const rect = u.rect(w, h, [-w/2, -h/2])
        models[pname] = p.position(rect)
    }
    return {models: models}
}
```

This is potentially due to the fact that the `perform_mirror` function simply does a deep copy of the original point, before applying mirror-specific changes. Ideally the new point should be run through the same calculations for non-mirrored points (`assert.sane -> assert.mathnum -> mathjs.evaluate`). A quick fix would just do this in the `visualize` funciton.